Cislunar-Explorers/FlightSoftware #43: NEMO: Replace RPIO.GPIO with pigpio

tmf97, 2021-03-17:
We are no longer using the RPI.GPIO library for controlling the GPIO pins, for timing reasons. 
I'm not sure how using two different libraries to control GPIO outputs would clash (if at all). If you have reason to believe it's not an issue - please let me know!

pigpio: http://abyz.me.uk/rpi/pigpio/python.html
`pip3 install pigpio`

The syntax between the two libraries are very similar.
With `pi = pigpio.pi()` 
`GPIO.setup(channel, GPIO.IN)` is now `pi.set_mode(channel, pigpio.INPUT)`
`GPIO.output(channel, GPIO.HIGH)` is now `pi.write(channel, 1)`
etc...

matthewlh, 2021-03-17:
Ok, I'll get the nemo code switched over to pigpio for GPIO access.

It looks like you're using both pigpio and busio for I2C, is there a preference which should be used for nemo?

tmf97, 2021-03-17:
I have a very slight preference for busio for Nemo's i2c because that's what most of our other sensors use. The only reason we're using pigpio for i2c is because of that specific driver's heritage.

matthewlh, 2021-03-17:
Copy that, I'll stick with busio then.

matthewlh, 2021-03-17:
Resolved in a5fad5a3c81f96b0495f382b54550d13f383a367

closed 2021-03-17 by matthewlh